CoRoSoN-Kit/CoRoSoN-Kit #2: select a motor and motor driver

tknobi, 2022-05-05:
Copied from discord and translated using deepl:

@stiebel :

> Just did a search for engines: https://eckstein-shop.de/V-TEC12VMini25DDCMotorGleichstromGetriebeMotorStirnradgetriebe770RPM
> Seems to be the Pololu 12VMP25D with 9.7:1 that we have been using for years.
> 8,50€ for metal gear is very cheap. I know Eckstein as a reliable store.
> To go with it, the Infineon IFX9201SG would be an ideal motor driver. 4 Euro, goes directly between controller and motor, no additional components needed, but SMD. We have in breakout - board for this : https://github.com/stiebel/bohlebots/blob/main/eagle/ifx9201breakout.lbr
> There we would have with 3 engines the complete drive train at under 40 euros. The question is if we want to allow SMD for this or even offer a populated breakout board. If you know a wired driver of the same quality, we will take it of course.

@Alintya : 

> maybe https://www.st.com/en/motor-drivers/l6207.html

@tknobi :

> There is this evaluation board for 2 motors to run the motor driver without SMD, but it is big and has a lot of stuff you don't need / want:
https://eu.mouser.com/ProductDetail/Infineon-Technologies/H-BRIDGE-KIT-2GO?qs=Y3VK4YDeAq%252B7x3331q9o4g%3D%3D

tknobi, 2022-05-17:
Before selection a motor / motor driver. Actual problems / points to be discuss: 
1. We have not yet decided whether to use SMD components or through-hole components. These would be much easier to solder for beginners.
2. There is a discussion by RoboCup officials to reduce the allowed voltage in the rules to 9V.
3. Do we plan a robot with 3 or with 4 wheels? Or do we support both?

tknobi, 2022-06-05:
In the last Discord meeting we (@Alintya, @f-gaul, @JBrutzi, @heikobaumann and @tknobi) briefly researched whether there is a motor driver that can be ordered already assembled (so that users do not have to solder SMD. see #15)

On [this](https://www.pololu.com/category/11/brushed-dc-motor-drivers) Pololu site we quickly found many drivers which seem to fit at first sight, e.g. [DRV8874](https://www.pololu.com/product/4035)

@f-gaul volunteered to do deeper research.

stiebel, 2022-06-06:
The DRV887a sound perfect to me.

dbscoach, 2022-06-06:
Wenn wir Code für Anfahren mit Gasstoß (Anlaufstrom und so) in den Libraries unterkriegen, die wir shippen wäre das vermutlich nicht schlecht für die Brushed-Getriebemotoren, weiß aber nicht wie elendig viel Tuning-Aufwand das wäre oder ob das überhaupt in zuverlässig geht... Wäre jedenfalls eine Vereinfachung für Teams, die bisher Lego-Bots ja auch mit "dreht so schnell wie man das eingibt" fahren können (wobei die das glaube ich mit Encoder lösen)