CoRoSoN-Kit/CoRoSoN-Kit #12: Finding a new name

tknobi, 2022-05-17:
We need a new name, because there is already a soccer-playing robot named [Bruno](https://robots.ieee.org/robots/dribbler/).
This is to prevent confusion.

* Deadline: Start of [European Robocup Junior] (https://2022.robocupjunior.eu/) (June 1st, 2022), as we want to present the concept to other teams and/or official there.
* We should use the term **KIT** instead of **ROBOT**. This is to clarify that we do not want to provide robot instructions, but only instructions on how to build components.
* The idea to use an acronym was generally accepted in the meeting, but some creativity is still missing :smile:

tknobi, 2022-05-17:
**CoRSN** kit:  **Co**mponents for **R**obocupJunior **S**occer **N**ewcomers

stiebel, 2022-05-19:
RoSoNeSt kit
Robocup Soccer New Standard

stiebel, 2022-05-19:
RoSoNeCo kit
Robocup Soccer Newcomer Components

tknobi, 2022-05-19:
**CoRoSoN** kit:  **Co**mponents for **Ro**bocupJunior **So**ccer **N**ewcomers

stiebel, 2022-05-19:
CoRoSoN sounds great to me!
It actually remembers me of Coruscant, the center of power in Star Wars...
That fits !

f-gaul, 2022-05-23:
Here are two more options:
1.) **Be**ginner **R**obocup **S**occ**er** **K**it -> BeRSerK

f-gaul, 2022-05-23:
2.) **B**eginner **R**obocup Jun**i**or **S**occer **K**it -> BRiSK

f-gaul, 2022-05-29:
Keep It Simple -> KIS

stiebel, 2022-05-29:
Hello altogether,
I have seen plenty of good ideas here.
My favorite is CoRoSoN, but I also like KIS, even if I would make it a
Keep It Simple Soccer -Kit -> KISS-Kit, although thinking about it, this could pop up in wrong queries...

However, we (you) should get a final name an Tuesday, so I can spread the good news to the (European) World 
from Wednesday on.

Greetings,
Roland

f-gaul, 2022-05-30:
I think
Keep It Simple Soccer -Kit -> KISS-Kit 
is fine too.

closed 2022-05-30 by tknobi