CoRoSoN-Kit/CoRoSoN-Kit #15: SMD components or through-hole components only?

tknobi, 2022-05-23:
We want to discuss here whether we want to use through-hole components only, or also SMD components, or only SMD components.

pros for through hole only

* For beginners, soldering through-hole components is much easier than SMD

pros for SMD also

* many components (e.g. motor drivers -> #2) are only available in SMD

Alintya, 2022-05-23:
We are _very_ likely to not come by every single part in THT. I therefore suggest not stressing too much about it. Basic components like resistors can be soldered quite well with little practice. We should just try to stay from really tiny SMD parts like 1005 if not absolutely necessary.

tknobi, 2022-05-31:
Just decided in Discord meeting together with @Alintya, @heikobaumann, @f-gaul , @JBrutzi: 

No SMD soldering for users of the kit. Maybe there will be mandatory exceptions at some point, because certain parts are simply just available as SMD, but we are trying everything to avoid that.

However, this does not mean that we do not use SMD parts at all. If parts can be bought already soldered (e.g. an assembled motor driver board) that is of course perfectly fine.

ToDo:
* [ ] Add decision to decision board.

PR #38 (2022-06-09): Update decision board and readme
resolves #16
resolves #15

just added the decisions from our last discord meeting
closed 2022-06-09 by Alintya