CoRoSoN-Kit/CoRoSoN-Kit #16: limit voltage to 9V ?

tknobi, 2022-05-23:
We herd that there is a discussion by RoboCup officials to reduce the allowed voltage in the rules to 9V.
So, to avoid later redesign, should we limit directly to 9 V?
This would have a particular impact on #2

tknobi, 2022-05-31:
Just decided in Discord meeting together with @Alintya, @heikobaumann, @f-gaul , @JBrutzi:

There is a German saying: 

> We are talking about unlaid eggs!

So, as long as there is no official rule decision to limit the voltage to 9 V, we also do not pay attention. Especially the motor selection would be much more difficult.

However, we can of course keep it in mind and use e.g. motor drivers that already work at 9 V.

ToDo:
* [ ] Add decision to decision board

stiebel, 2022-06-06:
having good motor drivers (5-20V or wider)  make this discussion obsolete.
There are plenty of motors around for all voltages.

dbscoach, 2022-06-06:
The idea was to slow the fast and strong robots down with a measure that is easy to implement for tournament organizers. 
Convincing arguments that this would slow down only the (financially) weaker teams if anyone at all were made here https://junior.forum.robocup.org/t/resolve-lowering-voltage-limits-to-9v-will-significantly-reduce-robot-speed-show-me-the-data/2279/ and accordingly the 9V rule is very unlikely to be coming, it was just already "in print" when these points were raised. 

Power limiting by means of an XT60-interceptor-board is now in discussions - though for 2024 at the earliest. 

Impacts also impacts #35 I think

PR #38 (2022-06-09): Update decision board and readme
resolves #16
resolves #15

just added the decisions from our last discord meeting
closed 2022-06-09 by Alintya