CoRoSoN-Kit/CoRoSoN-Kit #17: 3 or 4 wheels robot ?

tknobi, 2022-05-23:
Do we plan a robot with 3 or with 4 wheels? Or do we support both? see #2 

I think we should put as little restriction as possible here. I also think that the number and position of the wheels must remain team decision to comply with the rule "`Robots must be constructed and programmed exclusively by student members of the team.`".

What do you think?

f-gaul, 2022-05-23:
If the construction of the chassis is to be done independently by each team, then we should also remove the items 'Chassis' and 'Number of Wheels' from the decision board completely.

tknobi, 2022-05-31:
Just decided in Discord meeting together with @Alintya, @heikobaumann, @f-gaul , @JBrutzi:

We support both, it's a team decision.

Decision board already updated by @f-gaul in #19

closed 2022-05-31 by tknobi