CoRoSoN-Kit/CoRoSoN-Kit #18: Select a brain / main cpu

tknobi, 2022-05-23:
Basically there are 2 possibilities here:

1. We build our own board, take care of power supply, programming adapters, connectivity, IDE integration, etc. ourselves. This would give us full design control.
2. We use an existing single board chip, which is tested and takes a lot of work off our shoulders, but of course costs us some flexibility.

Should we find a suitable board for solution approach 2, I would personally be strongly in favor of it. 

Therefore, here is a tabular comparison of boards which we had already considered in verbal discussion or in Discord:

*Work in progress!*
 
| Property | Raspberry Pi Pico | ESP32 DevKitC | Arduino | WeAct Black Pill V1.2 | `? Your Proposal` |
| --- | --- | --- | --- | --- | --- |
| Supplier |
| Open Source |
| Dimensions |
| Price |
| Availability |
| CPU |
| RAM |
| Flash |
| No of digital I/O |
| ADC |
| I²C |
| SPI |
| Buttons |
| LEDs |
| Arduino IDE integration |
| ? |

stiebel, 2022-05-31:
No of io Pins 
This depends. We startet with a 40$ teensy once. To get more Pins we used i2c pirt expanders and multiplexers.
But with a 4$ pi pico we would instead simple use two of them.
So for lets say 12 - 24 Pins for IR Sensors is is the cheapest and easiest way to use a Single pico and make the whole circuit Board an "ir ring sensor with i2c"

stiebel, 2022-05-31:
Please enlarge the list:
Can Bus
Bluetooth 

The esp32 does both buildin, which is its highest advantage above the pico as far as i am concerned

stiebel, 2022-06-06:
I absolutely understand the wish for 5V tolerance, but all these modern and cheap CPUs like ESP32 and Raspberry Pi Pico are NOT 5V tolerant. So I would rather vote for a "lego-adapter" ore even a "lego-shield" to adopt to 5v envirement,
Apart from lego there is only the pixy-cam that wants to have 5V for power but has a 3.3v Level UART/SPI/I2C communication.
all other Stuff should work with 3.3v

dbscoach, 2022-06-06:
Built-in level shifters (for some 5V-capable ports) and/or protective diodes so you don't accidentally kill the board are a completely valid substitute, I agree. I just think we should design for as much robustness as we can (within reason) - or we can just have a socketed chip/subboard because replacing a 5$ chip is a valuable learning experience in and of itself...

tknobi, 2022-06-07:
Decision from the last discord meeting together with @f-gaul , @heikobaumann, @JBrutzi , @Alintya & @tknobi  : 
We use one (or more) single board chip(s). This is the only way we can use a modern chip without soldering SMD. Besides, these are now so established and mature that you would have no advantage to do it yourself. Let's finish the comparison table.

ToDo: 
* [ ] update decision board

markosko, 2022-06-07:
For arduino maybe Arduino Mega PRO or Arduino Mini Pro?

Alintya, 2022-06-07:
The mini pro is really limited in I/O.
Mega Pro on the other hand is super rich in I/O and might be worth consideration on signal-heavy components like the (ir) sensor board.

tknobi, 2022-06-08:
> For arduino maybe Arduino Mega PRO or Arduino Mini Pro?

Thanks @markosko for your proposal! Can you share a datasheet or a product page? On the official [Arduino Pro website](https://www.arduino.cc/pro/hardware/product-family/portenta-family?id=25729177) is no Mega or Mini. Or are you talking about an unofficial version?

markosko, 2022-06-08:
Yes, those unofficial /cloned but except of layout of pcb, pins are same

Alintya, 2022-06-09:
Mostly updated OP for the Blackpoll board

tknobi, 2022-06-14:
Decided in meeting with @f-gaul, @Alintya, @stiebel, @markosko and @tknobi:

We don't want boards that are just imported from China or don't have an official manufacturer. Therefore, we exclude the WeAct Black Pill and the Arduino MEGA 2560 Pro.

Of the well available remaining ones, the ESP32 is the one with the most features, but also the most expensive. So we will try the Raspberry Pi Pico. If this does not work (e.g. because only 3 ADC's we will think again about the ESP32). 

ToDo to close issue:

* [ ] update decision board

PR #45 (2022-06-28): Update docs
resolves #18 
resolves #11
closed 2022-06-28 by f-gaul