CoRoSoN-Kit/CoRoSoN-Kit #20: How much I/O-Pins does Brunos CPU really need?

f-gaul, 2022-05-29:
In order to decide which CPU to use, we need to determine a minimum number of I/O pins required. I'll start with that:

* motor driver: 3-4 motors, each 1 pin power(PWM) and 1 direction(I/O)
* kicker: 1 pin I/O
* ball sensor (ball in kicker) 1 pin I/O
* line sensor: 1-4 analog input
* distance sensor: 2 analog input
* ball sensor: 1 analog input
* mode switch (forward/defender) 1 I/O
* status LEDs: 2 I/O

**total:**  
* 4 PWM  
* 9 I/O  
* 4 anlog In

This means an Arduino Nano or similar is possible but already very tightly dimensioned.

Ok, what do you think about that...

tknobi, 2022-05-29:
> distance sensor: 2 analog input

why analog input? and why 2?
I think we should use 4 (each direction). And for ultrasonic you need a trigger and an echo pin: 1 trigger + 4 echo = 5 I/O

> mode switch (forward/defender) 1 I/O

In my experience, you can never have enough switches :laughing: , so maybe add at least 2?

Additionally we should provide:

* 1 PWM and 1 direction pin for a potential dribbler.
* an option to connect a display, which could be connected via:
  * Datapins:  6 Pins: (Reset, Enable, 4 Datapins)
  * SPI: 4 Pins: (MISO, MOSI, CS, CLK)
  * I²C: 2 Pins (SDA, SCL)

We might also want to include additional interfaces to make it easier for teams to create their own individual enhancements. Candidates here would be SPI, I²C, UART, simple I/Os.

Which CPU meets the requirements set up here should be discussed in #18.

f-gaul, 2022-05-29:
My idea was to start the discussion with the minimum I/O pins needed so that we can move forward from this point to push Bruno the next step to the real world.

> distance sensor: 2 analog inputs

In my mind we need a minimum of two distance-sensors to determine the distance to left and right (or rear/front) simultaneously. 
I was thinking of a sensor in the style of the Sharp-Sensor. (I know that the Sharps are IR-Sensors and that's not allowed to emit IR-light). But, I think it should be possible to build or buy a none IR-Distance-Sensor which works the same way and has a voltage output according the measured distance.

I absolutly agree that we should have 4 distance sensors and that they are not using IR-light. This discussion should clarify the 'must have' and the 'nice to have' things and we are on a good way to find out.

So far we have:

* motors -> 4 PWM and 4 I/O
* distance -> 5 I/O
* kicker -> 2 I/O
* dribbler -> 1 PWM and 1 I/O
* line sensors -> 4 I/O (or analog input)
* mode switches -> 2 I/O
* start button -> 1 I/O
* status LED -> 3 I/O
* SPI -> 4 I/O 

total: 5 PWM and 26 I/O

Things are already becoming clearer. We left the 'Arduino-Nano' region and know that we need something with more I/O Pins then 20. Let's go on...

f-gaul, 2022-11-01:
Since we have decided to use a Pi-Pico this issue can be closed.

closed 2022-11-01 by tknobi