CoRoSoN-Kit/CoRoSoN-Kit #22: Select an orientation sensor

tknobi, 2022-05-30:
Should we use a magnetic sensor, or an gyroscope or an accelerometer? 
To be honest, I'm not up to date with these sensors in the RobocupJunior Soccer context.

@stiebel , @heikobaumann: Can you recommend a usable, cheap and readily available sensor?

stiebel, 2022-06-06:
We use the cmps14 (which uses a BNO Chip) and the original BNO55.
Both are ok but nor perfect.
Sensors have a tendency to rater drift (the old CMPS11) or jump (+-20° at once) (all BNO chips I know)
Using a compass without gyro is a bad idea because of the metal constructions underneath the tables.
I do not know if there is any perfectly working sensor on the marked...
But: in 1vs1 a 20° jump is not fatal and you can press the "calibrate" - button after every goal.

So if there is no better choice I would go for the 30€ BNO55

dbscoach, 2022-06-06:
Most teams except Standard Kits seem to have gone to IMUs with some sort of compass/gyro fusion - some of them not calibrating correctly which then leads to a jump when the robot has made enough turns to be calibrated to true magnetic north - so we need to choose something we can explain very clearly and maybe even take steps so it doesn't turn off easily/accidentally (like keeping it powered as long as battery is present and only reset by reset-button) so teams don't fall victim to "use what I don't really understand"-risk there.

markosko, 2022-06-06:
For BNO55, calibration could be done after every power up with few turning or making few movements in air by hand to have sensor be calibrated. And should be stayed same way before power down/up cycle resets memory. With program made in way that robot starts with program or some kind of pause in program with waiting for buttons or something similar.