CoRoSoN-Kit/CoRoSoN-Kit #23: User Interface Selection

tknobi, 2022-05-30:
I think, our user interface should provide:

* min. 2 switches (you can never have enough)
* min. 1 button (to start the robot at kick off for example)
* a display for debugging opportunity
  * how large should it be?
  * text only or also images?
  * is rgb needed?
* *Not sure / optional*: 
  * a rotary encoder
  * a buzzer / a sound module 
  * touch screen

To discuss:

* What do you think? 
* Is something missing? 
* Is something not really needed?
* Should it be connected to brain / main cpu directly or should it have a own processor?
* Should it be always connected or plug and play with a cable (to save weight)?

stiebel, 2022-06-06:
If you have 4 buttons and 4 RGB LEDs in my opinion there is no need for a display.
You can realize this very easy with a I2C 8bit-Port expander. 2x3 bit for 2 RGB-LED -> 8 Colors and 2x1bit for 2 Buttons.
This can be done as a module to place wherever you want...

dbscoach, 2022-06-06:
Teams can always upgrade by hooking up an SPI LCD module of their choice, that would only need one IO as chip select for the screen and could use existing SPI otherwise if we don't have any spares on the MCU.