CoRoSoN-Kit/CoRoSoN-Kit #24: Select line detection sensors.

tknobi, 2022-05-30:
We need to select sensors that reliably detect the line of the field.
In my active time there were no lines, so I lack experience here.
 
The following questions need to be answered (I think):

* Which sensor should we use?
* How are the sensors read out?
* How many do we need?
* How should they be arranged? In a circle? As a cross? Different?
* Should a separate processor do the evaluation and provide the result via a bus or should the sensors be connected to the main processor?

Do we have an expert here who could answer the questions?

stiebel, 2022-05-31:
Do we really need line detection in 1vs1?

tknobi, 2022-05-31:
> Do we really need line detection in 1vs1?

No, not in the 1vs1 league. But I think it should be possible to use our kit in every league.

My proposal is, we care about line detection later in one of the next versions and ignore it in the initial minimal release.

f-gaul, 2022-05-31:
A very simple line-detector could be built with a LED, a fototransistor and two resistors. If such a device is placed in the middle of the robot it is possible to detect lines bevore the robot match the 'Out of Bounds' condition. Only one analogue input is needed for this.

dbscoach, 2022-06-06:
If you want to make it simpler for teams a bigger line sensor (cross or circle) would be nice because it is possible to know where the line is coming from with less complicated logic. If you have just the one you're out of luck when you a) get pushed over because it's not pushed out if the robot is alone by the time it crosses the line and b) have a hole in your logic for a weird edge/corner case (esp. relevant now that the penalty area is a weird-shaped white line). Even if we put it in just one space I'd suggest having at least a couple phototransistors so you can know what direction you're going and detect wether you went in-field->on-line->in-field or in-field->on-line->out-of-field.

markosko, 2022-06-08:
For phototransistor my team tested  SFH309 after getting reference from one other team
About cross or circle it depends circle has nice vector calculations for getting out of line, cross is kind of easier for younger kids probably. For cross at least better if there would be two leds and two phototransistor. With 8transistors not that much of additional chip to make kidn of module but direct connection to main chip could be used too