CoRoSoN-Kit/CoRoSoN-Kit #26: Omniwheel: Print one or select one to buy?

tknobi, 2022-05-30:
An Omniwheel drive has become the defacto standard in RoboCup Junior Soccer in recent years.

1. There are ready-made Omniwheels with which many teams have already gained experience. We should possibly make a list of them here and explain why they are (not?) suitable.
2. More and more teams are building omniwheels themselves. We would like to offer a completely 3D printed Omniwheel, as this would be easy and cheap for everyone to manufacture.

stiebel, 2022-06-06:
I would love to see working 3d printed wheels.
(I have seen some not working in Portugal last week)
@tknobi : can you provide STL file for test printing your solution?

dbscoach, 2022-06-06:
A comparison chart would be a good resource even for the non-beginners, I started a forum thread here: https://junior.forum.robocup.org/t/omniwheels-comparison-tell-us-about-yours/2491 . We did see mostly bought omniwheels (except Entry of course) in Portugal last week with the self-built ones mostly being old stock from previous teams as far as I can tell (we asked around but not get to quite everybody). 
There was one team that had designed and printed their own omniwheels, they struggled with stress concentrations around the axles and said to use the largest hardware you can around mounting points. I worry about durability for beginners running those.

To get the comparison chart started: One of the sources people use is https://gtfrobots.com/product-category/omni-directional-wheel/ , there is another one I'll probably get the link for in a bit, do we keep track here or on the forum where teams are more likely to give input or both?

tknobi, 2022-06-09:
> @tknobi : can you provide STL file for test printing your solution?

I must correct, I did not create one and never printed one. But I know that @fabiomanz from `The Rockys` did. He has also agreed to publish it and will probably do so here soon. :tada:

fabiomanz, 2022-06-11:
I added the files in the following PR: #39