CoRoSoN-Kit/CoRoSoN-Kit #27: Find an IDE integration for textual programming

tknobi, 2022-05-30:
A robot built with our components should be easy to program.
Besides the easiest way of visual programming, we should also provide a textual IDE.

I see 2 approaches here that I would classify as simple enough: (do you know more?)

1. Arduino IDE, programming in C.
2. MicroPython: I haven't used it (yet), but it's supposed to be easier for beginners than C. I don't know which IDE is suitable yet. 

With both approaches, I wouldn't know how to integrate custom components. Does anyone have experience?

Alintya, 2022-05-30:
I vote for providing an arduino package. the API is simple enough and their boards are already widely spread. (so people will be familiar with the IDE anyways)

stiebel, 2022-06-06:
I agree, but only because I do know arduino much better than micropython.
However, this does not have to be an "or" Question.
Using the Pi Pico or the ESP32 allows both.

f-gaul, 2022-07-03:
I have not worked with 'MicroPhyton' yet either. In my opinion, the Arduino IDE would be a good choice. It offers learning groups the possibility to start with a simple board e.g. Arduino-UNO and to gain first experiences with microcontrollers and electronics. Thereby you can see if you like it and you can also write simple, small C-programs and thus gain practice in programming.
Possibly there are also already working groups which have begun in such a way. If the enthusiasts then decide: 'Let's build a soccer robot', you can simply continue and expand on what is already known.

Translated with www.DeepL.com/Translator (free version)