CoRoSoN-Kit/CoRoSoN-Kit #28: How to make visual programming possible?

tknobi, 2022-05-30:
From personal experience (as a learner and teacher) I know that graphical programming is much easier for children than textual programming.
Therefore, we should make this possible.
I think it should be an open source solution, as we will probably have to customize it to give our users a good user experience. 

I could find the following canidates so far: 

* **BIPES**: Block based Integrated Platform for Embedded Systems: [Website](https://bipes.net.br/wp/), [Repo](https://github.com/BIPES/BIPES), [image], [image]
* **xod.io**: A visual programming language for microcontrollers: [Website](https://xod.io/), [Repo](https://github.com/xodio/xod), [image], [image]
* **ArduBlock** is a Block Programming Language for Arduino: [Repo](https://github.com/taweili/ardublock), [image], [image]
* **BlocklyDuino** is a web-based visual programming editor for arduino: [Repo](https://github.com/BlocklyDuino/BlocklyDuino), [image], [image]
* **BlocklyDuino v2(020)**, reboot and up to date with Google Blockly, graphical programming of Arduino boards: [Repo](https://github.com/BlocklyDuino/BlocklyDuino-v2), [image], [image]

Do you know others, do you have experience with one of the mentioned?

mrshu, 2022-06-08:
@tknobi I do have experience with Blockly, which is an awesome platform to build upon.

That being said, what sort of age categories are you aiming at in here? That would help clarify the actual choice in the end :slightly_smiling_face:

Alintya, 2022-06-09:
@mrshu I personally expect coroson to be viable from ~7. grade up (12+ years old).
Higher grades, let's say beginning at age 15, should move towards code.

tknobi, 2022-06-09:
> @tknobi I do have experience with Blockly, which is an awesome platform to build upon.

@mrshu Great to have someone with Blockly experience here. :rocket: I think no matter what solution we will have to create our own blocks (e.g. an IR sensor block).

> That being said, what sort of age categories are you aiming at in here? That would help clarify the actual choice in the end :slightly_smiling_face:

Honestly, I think that a visual programming environment does not necessarily depend on age, but rather on programming experience. I've met enough adults who preferred to learn programming visually rather than textually.

Alintya, 2022-06-09:
> > @tknobi I do have experience with Blockly, which is an awesome platform to build upon.
> 
> @mrshu Great to have someone with Blockly experience here. 🚀 I think no matter what solution we will have to create our own blocks (e.g. an IR sensor block).
> 
> > That being said, what sort of age categories are you aiming at in here? That would help clarify the actual choice in the end 🙂
> 
> Honestly, I think that a visual programming environment does not necessarily depend on age, but rather on programming experience. I've met enough adults who preferred to learn programming visually rather than textually.

Do we artificially limit the possibilities of the visual environment then? (most other frameworks do, unclear [to me] whether due to cost or intention)
I do think age is bound to the choice of programming environment. Or do you see a 6th grader spitting out c++ casually?
It is clear to me that the entry point especially for younger students should be the visual route (unless they are somehow super experienced with code).
Question is, do we want code to be the natural next step in advancing or just a choice?

tknobi, 2022-06-14:
I just added mBlock to the list since @stiebel had good experience with it.