CoRoSoN-Kit/CoRoSoN-Kit #32: [question] Involvement of target group

JonGitti, 2022-06-03:
This is my first comment here.
Ex Participant, Ref, experience (in failure) to pass technology and knowledge to the younger.

In general I'm willing to contribute.
I am sure you can develop the kit as it is no rocket science and we know the requirements well.
However, there are a lot of projects from great people with great intention which are dumped in the end due to non-technical issues such as wrong expectations, lack of involvement of the target group and so on. I want to be convinced it has a chance to actually come to use and boost children's learning progress.
**Are there connections to schools/organizations supporting this idea who will be candidates for launch?** I think it could be crucial that they develop an intrinsic motivation to use the kit. Through an exchange of ideas and issues.

Is GitHub the platform to use for such kind of non-technical stuff?

stiebel, 2022-06-03:
Hi and welcome!
You ask about schools involved.
Yes, this is probably my main part in this project.
I will start to try our ideas of the coroson kit with my students in september to take part in the robocup season 2023 and i try to get more schools to get involved.

JonGitti, 2022-06-03:
ah, bohlebots. Sorry didn't see you are involved.
maybe the existing network should be visible somewhere

tknobi, 2022-06-07:
Welcome @JonGitti :wave: 
 
> I am sure you can develop the kit as it is no rocket science and we know the requirements well.
However, there are a lot of projects from great people with great intention which are dumped in the end due to non-technical issues such as wrong expectations, lack of involvement of the target group and so on. I want to be convinced it has a chance to actually come to use and boost children's learning progress.

True words, thank you! We try everything to avoid that here only a few "robocup alumni" play again and the components finally no one uses.

> Are there connections to schools/organizations supporting this idea who will be candidates for launch? I think it could be crucial that they develop an intrinsic motivation to use the kit. Through an exchange of ideas and issues.

Besides coach @stiebel from [Bohlebots](https://www.bohlebots.de/statisch/index.html) also @f-gaul and @heikobaumann from [The Rockys](https://www.robotik-ag.eu/) are on board. 2 teams are of course only a beginning and not enough to really establish the components, which is why we cordially invite interested teams or coaches to participate.

>  maybe the existing network should be visible somewhere

you are right, i opened issue #36 for it

> Is GitHub the platform to use for such kind of non-technical stuff?

Sure it's fine, if it gets to "off-topic" you are also invited to chat with us on [image].

> This is my first comment here. In general I'm willing to contribute.
Ex Participant, Ref, experience (in failure) to pass technology and knowledge to the younger.

Great! Would love to learn from your experience. We are in a video call (also on Discord) every second Tuesday. You (and anyone else interested) are welcome. Next meeting is at June 14, 08:15 pm (CEST)

closed 2022-06-08 by JonGitti