CoRoSoN-Kit/CoRoSoN-Kit #34: Do we need a camera?

tknobi, 2022-06-05:
This issue was created in the Discord meeting with @Alintya, @heikobaumann, @f-gaul and @JBrutzi:

We have not yet decided whether to offer a camera module.
Some teams use the also open source camera [OpenMV](https://openmv.io/), maybe we should only provide control to this and recommend it?

dbscoach, 2022-06-06:
At least as popular as the OpenMV is the PixyCam (also Open Source it seems) https://pixycam.com/source-code/ .

dbscoach, 2022-06-06:
If you're planning for this to be usable in Open League (which for many countries with less established RoboCup presences isn't as far fetched as it sound for e.g. Germany, we did see Lego Bots in LWL in Portugal and in OL in Sydney - although the latter didn't work) or a proposed cost-down Entry league (like Mike suggested with bigger orange balls in the forum https://junior.forum.robocup.org/t/coroson-kit-components-for-robocupjunior-soccer-newcomers/2484/2 ) you'd definitely need one. An important consideration may become the fact that fisheye lenses and multiple cameras will most likely be legalized in 2023 because the mirror/lens-DIY-requirements did the opposite of what they were supposed to (idea was that buying up to 3000€s worth of mirrors was to be disallowed but it lead to only teams which can get a hold of custom manufacturing somehow being able to have decent optics at all, esp. with just the one camera).

markosko, 2022-06-13:
In lightweight categories camera is nice to have because compass will help with orientation but not where goal. As same value for compass for rotation is from left of goal, at center of goal and right from goal.

JonGitti, 2022-06-14:
[From the pixy2 docs](https://docs.pixycam.com/wiki/doku.php?id=wiki:v2:porting_guide#notes-on-each-interface)

> Note, all digital output signals on Pixy2 are 3.3V CMOS logic. All digital input signals on Pixy2 are 5V tolerant.

>SPI
3.3V outputs, 5V tolerant  inputs
I2C
>There are weak 82K pullups to 5V on SDA and SCL signals, via R14 and R15.

They changed it from an [early version](https://docs.pixycam.com/wiki/doku.php?id=wiki:v1:Porting_Guide#notes-on-each-interface) where it was 4k7. Is it really a dirty hack or even intended to connect to 3,3v controllers @stiebel 's way?

Alintya, 2022-10-19:
With mindstorms slowly fading out and the future of the IR-ball being very uncertain beyond 2023, it is quite clear that we are going to need a camera module even for non-open leagues.
Let's move this to discussions #49

closed 2022-10-19 by Alintya