CoRoSoN-Kit/CoRoSoN-Kit #35: Power supply brainstorming

tknobi, 2022-06-05:
This issue was created in the Discord meeting with @Alintya, @heikobaumann, @f-gaul and @JBrutzi:

So far we have not discussed how our components are powered.

* Which battery type do we use?
* Do we need a fuse?
* Do we need voltage regulators?
* Are these things on the main board or separately?
* How to connect the battery?

dbscoach, 2022-06-06:
As a battery connection I'd argue for XT60 because:

- Widespread already (de facto standard in all but two teams at RCJ Europe last week, similar at German Open and at least widespread in other super-regions to use RoboCup-speak)
- Rated for 1000 plug/unplug cycles
- Forward compatible for proposed (2024+) power measuring/limiting bewteen-battery-and-bot-board

On voltage regulators I'd argue for simplicity, we should be able to get by without providing #16 goes the way I think it will.

Fuse is too cheap not to do for newbies, maybe even an elecronic so they don't need to wait to buy replacement parts if something goes wrong. 
Undervoltage protection would be good for newbies if we use lithium anything, even if it's just putting one of those beepers that go into the balancer port on the BoM. Maybe a consevatively set voltage cutoff on the main board could be useful in case people don't plug in the beeper.

All this would be assuming LiPo/LiIon off the shelf packs which I think are a reasonable enough solution (cheap, durable enough if you don't mistreat them, lightweight).

stiebel, 2022-06-06:
I agree to all!
I think also  very important is a reverse polarity protection.
Even if we want to use XT60 you can still use them wrong

Alintya, 2022-06-07:
> I agree to all! I think also very important is a reverse polarity protection. Even if we want to use XT60 you can still use them wrong

I came across that argument multiple times over the years during my RC adventures and I still wonder how on earth would one manage to plug an XT60 the wrong way? The sheer force needed to do is immense.

markosko, 2022-06-07:
Plug on its own has only one way to plug it, but can be badly soldered

Alintya, 2022-06-07:
> Plug on its own has only one way to plug it, but can be badly soldered

Well, you can do that to literally anything. Considering that the reverse polarity protection has to be soldered as well in the first place, there will be just as many ways of messing up.
Coaches should strongly encourage multimeter testing before plugging anything in anyways.