CoRoSoN-Kit/CoRoSoN-Kit #36: Show our network

tknobi, 2022-06-07:
Based the idea from @JonGitti in #32 we should possibly show our network in more detail. Who is collaborating? Which former and active robotics teams are represented? Which roles are present? (Coach, referee, participant, former participant, ...).

What do you think?

Alintya, 2022-06-07:
Where would that help?
Shouldn't affiliation be rather be rather irrelevant to the project?
I mean we could add some sort of 'experience level' or field of expertise to the discord at least such that people have a better understanding of who to talk to over specific topics.

dbscoach, 2022-06-07:
"XY school/institution which is known for successfully getting their students into robotics is involved in developing/uses/plans to use this platform" is probably a strong argument for teachers making platform decisions - they largely don't have the time and resources to undertake detailed technical evaluations or to take a big risk on a new platform they aren't sure about. 
What they need is some assurance that spending limited resources on this will help get their students build working robots and have learning experiences. Therefore I think we should be willing to advertise who we are/who we are working with as appropriate (and with previous coordination and so on).

JonGitti, 2022-06-08:
@Alintya My issue is not about the affiliation of individuals, and not about who can I talk to with expertise (public, sb will assign), but about the chances of making the project's vision real. I see a repo in the initial stage. "Does the project have good plans how to get this rolled out?" people landing on this page might wonder (and I did). I want to get involved only if the project has chances not visions only; I don't want to create sth cool here, but foster actually kids' learning process in rcj. Achieving this goal is equally or even more dependent on the connection to its users than on its developers i.m.o.

This comment explained #32 , a question left open to newcomer devs. It does not address marketing to the user group as the comment above, agrees implicitly for the need of it though.

JonGitti, 2022-06-08:
> Based the idea from @JonGitti in #32 we should possibly show our network in more detail. Who is collaborating? Which former and active robotics teams are represented? Which roles are present? (Coach, referee, participant, former participant, ...).
> 
> What do you think?

to clarify. faces, personality and background of collaborators are not important and I didn't address this.
Showing the presence of roles could be helpful but it is actually already done in the description so don't bother with it too much.

Alintya, 2022-06-08:
Isnt that kind of a hen and egg problem though? People don't want to put their work into an unpromising project and the project will never look promising without work being put into it.
But I agree to a certain degree. We should extend on our future plans in the readme and show off some teams involved like bohlebots.

JonGitti, 2022-06-08:
> We should extend on our future plans in the readme and show off some teams involved like bohlebots and other worthy mentions of involvement going forward.

Good.

@Alintya ok I also had the chicken-egg moment while typing the previous comment, but came to the conclusion it is none since you can make the project look promising to the organizations also without a lot of devs.
Then, when it's known that interested users are there, aware, waiting or just following this project, it looks more promising to put work into implementation.
The latter depends on the first, but the first only on you. So not that much a mutual dependency where my way of thinking hampers growth.

Alintya, 2022-06-09:
Shall we add a `Why we will succeed` or `Supporters` section to the readme then, laying out who is involved?