CoRoSoN-Kit/CoRoSoN-Kit #40: Connectivity concept

JonGitti, 2022-06-12:
What about the connectivity concept in general?

- Are participants expected to care about pinout and bus systems? Soldering, pin to pin wires from sensors to the main board?
- Or do we plan to wrap it and provide a generic sensor/motor plug like NXT did?
- How should our custom modules be connected (eg. IR-Seeker module, UI module)?

I mean not the exact system is relevant now, just the philosophy we want to pursue.
If this is answered somewhere else, you can reference and close.

markosko, 2022-06-12:
Hard to tell because it kind of depends on what would be mainboard, and even motors / other sensors  and so can have their own connectors so hard to tell.

Alintya, 2022-06-12:
General plan is to connect the bigger PCBs via SPI or I2C, for which we will provide a connector (that can easily be extended in itself by just daisy chaining)
Motor drivers and CPU boards will be plugged into generic pin strips

JonGitti, 2022-06-12:
For my question, with respect to the educational purpose and usability,
> not the exact system is relevant now, just the philosophy we want to pursue.

____
From your answers it seems like

> participants are expected to care about pinout

and we do not 

>  plan to wrap it and provide a generic sensor/motor plug like NXT.

Right?

____
_ok, example: suppose STM32-based mainboard with pin strips;
participants directly connect to its pins with no additional distributor/adapter?
the SPI connector we provide can thus still only consist of multiple females at the mainboard end, no easy single plug, or didn't I get it?_

tknobi, 2022-06-13:
> What about the connectivity concept in general?

Thanks for your inquiry @JonGitti.

>  If this is answered somewhere else, you can reference and close.

So far we have briefly touched on this in the meeting but nothing has been defined or documented yet.

--- 

> From your answers it seems like...

Sorry, but I don't read everything that way, I'll try to answer your questions a little more explicitly here:

--- 

> Are participants expected to care about pinout and bus systems? Soldering, pin to pin wires from sensors to the main board?

I think newcomers should not need to have a deeper understanding of bus systems to start with our components, similar to LEGO Mindstorms. However, this does not mean that they cannot learn it with the help of our documentation. For example, in the second year of participation, when they want to customize a component to gain a competitive advantage.

> Or do we plan to wrap it and provide a generic sensor/motor plug like NXT did?

Yeah that sounds more like what I had in mind so far. Whether there will be only one connector or several (one for motors, one for I¹C sensors, one for SPI sensors, etc.) still depends on some design decisions as already mentioned by @markosko  and @Alintya.

> How should our custom modules be connected (eg. IR-Seeker module, UI module)?

I think also with a bus system, so similar (or the same) as other sensors. I would like that a team with its own board can also use only one of our components. (e.g. the IR module)

> ... participants directly connect to its pins with no additional distributor/adapter?

No, I think there should be a board on which a single board cpu (e.g. a pi pico) is connected by a connector strips and which then provides connectors for sensors and motors and addional needed electronics (e.g. supply)

JonGitti, 2022-06-13:
thanks, this matches with my considerations.

> Sorry, but I don't read everything that way

The decision to encapsulate components is not dependent on which we choose, that's why I concluded from the prev answers ("depends on mainboard", "have their own connectors") that such wrapping is not planned.
>No, I think there should be a board on which a single board cpu (e.g. a pi pico) is connected by a connector strips and which then provides connectors for sensors and motors and addional needed electronics (e.g. supply)

But apparently it is planned, which I welcome. It is relevant because it determines what the kit should actually be, an _encapsulated but extendable system_ or merely a _collection of compatible components_. It's ok if you think _hollow words_, I think it affects the usability and learner's perception of the technology they work with. Thanks again for clarifying.

Alintya, 2022-06-13:
We definitely intend to be extensible as much as possible and I expect to provide additional connectors for 'leftover' interfaces we did not use in our basic design.

JonGitti, 2022-06-13:
yes that's clear but the design concept was not clear (and nowhere mentioned afaik) - that every ready-to-use 3rd party component will get its own adapter/integration board, e.g. that we design a larger pcb with a socket for anything like a stm32-blackpill, where the pcb will then provide all the interfaces just as we like to define them. No "wire yourself, RTFM", although still possible.

tknobi, 2022-06-13:
Yes @JonGitti it's nowhere mentioned yet. So before closing this issue we should:

* [ ] discuss about the connectivity concept in our next discord meeting tomorrow
* [ ] decide how we want to do it
* [ ] document it in our decision board

JonGitti, 2022-11-02:
Resolved by design proposed in #46.
I mean the hardware interface (=connectivity) concept, like wrapping buses in generic plugs and providing custom boards for each component.

closed 2022-11-04 by Alintya