CoRoSoN-Kit/CoRoSoN-Kit #44: Design a main board

tknobi, 2022-06-22:
After selecting a Main CPU in #18 and discussing about the required I/O's in #20 we can brainstorm about the main board design here.

Required components:

* Connector strips for a PI Pico
* Battery connector
* Voltage regulator
* Battery monitoring?
* fuse?
* 5x Connections to the motor boards. (3.3V, GND, PWM pin, direction pin)
* (as mach as possible) x Connections to sensors / sensor boards (e.g. IR module)
  * My suggestion here would be to try to build a universal connector. So one where pins for I²C, SPI, UART (and ADC?), general I/O are available. This would have the advantage that in the future every extension can use the protocol of its choice. (TODO: Check if this is possible, how big the connector will be and how many sensor connectors would be usable this way). 
  * An alternative would be to use only one communication interface (e.g. I²C).
* 2x switches
* 1x start button ? (not sure if we need it one of the switches should be used)
* 2x status LED ? (one led is already on the pico, not sure if we need 3 leds)
* Measuring pins for battery voltage, GND and 3.3 V

What did I forget?

markosko, 2022-06-22:
SPI, can be used for multiple connections but for that you had to need 3 pins + number of spi connections with power and ground pins, in case of three or more.
I²C is probably most expandable in case of multiple sensors if they have different I²C address, or other expansion boards and use only two pins + power and ground.
UART hardware wise is two pins per connection and there is only two pairs. There can be used software UART connections too.

But best to takeout and make connectors at certain spots to easier connections for each connection or even some connections for multiple power and ground pins 

About switches / start button, switch for power on / off states for motors. So for example there could be running robot so there could be way to check for info around like line, IR sensors and so, but without motors and it wont take input pins for settings.

Battery voltage can be checked with this or even plugged in while playing in case of  using Lipos, and there is space to store it inside of robot
[image]

And maybe in case of design try to check if everything can be squeezed in 18cm diameter ?

And maybe put two voltage regulators in case of 5V at same time as 3.3V ? Probably logic shifters too in case of using 5V and 3.3V sensors/modules ? Or at least put info about using them ?

tknobi, 2022-07-01:
I have a first draft of the brain pcb schematic (see #46)

Advantages:

* 2 switches and 1 button on board
* 5 motor connectors
* 8 universal "sensor" connectors
  - I²C on each connector
  - SPI on each connector
    * only SENSOR 5,6 & 7 have hardware CS (chip select) pin, but it should be possible to manually do chip select via software on every pin ([source](https://forums.raspberrypi.com/viewtopic.php?t=322617)).  
  - SENSOR 0,1,2 could be used to connect ADC sensors
  -  "Sensors" would have to be renamed, I am still unsure, possibly "components" (but no motors...).

Drawbacks:

* only 1 onboard  LED on Pi Pico
  - more could be connected externally to a "sensor" connector, since there is one I/O on each connector
* If someone wants to connect an I²C sensor, 
  - he needs a 10 pin connector, although he would need only 4 pins.
  - we waste an I/O, since this would then be unused at this sensor connector
* no UART connector
  - if someone really needs UART, it could be picked up at the IO pins of SENSOR 4 + 5

[image]